tjfenwick/DIY-Sim-Racing-Active-Pedal #1: ADS1256 Load Cell readings fluctuate and cause the motor to oscillate

tjfenwick, 2023-04-14:
The readings I am receiving from the load cell through the ADS1256 fluctuate around 0.5lbs. This causes the motor to oscillate when trying to hold a constant pressure on the pressure. I have mitigated this by adding a small dead zone on the pedal position that requires a certain number of steps before it will move along with a moving average on my load cell readings to try to smooth it out. The combination of these two has greatly reduced the oscillations but has added a slight lag to the pedal movement.

ChrGri, 2023-04-22:
Thank you so much for this awesome project.

I'm thinking about implementing a Kalman filter to estimate the load from the noisy ADC readings. It might work better than the MA filter, since motion assumptions could be integrated. Perhaps, worth a try.

Alternatively, when two (or N) loadcells are used, their individual reading could be averaged. Which, probably, would reduce noise.

BR

kiss81, 2023-04-23:
Definitely going to build this as well! Thank you for sharing! I was thinking if smoothing the input signal with a small capacitor.  But make sure to read this:
https://electronics.stackexchange.com/questions/218208/smoothing-load-cell-readings-with-a-capacitor

tjfenwick, 2023-04-24:
I am playing to switch over to a 100kg load cell (link below) like you suggested.

I am also plan on trying a ADS1115 or a ADS1015 to see if they work better.
 https://www.ebay.com/itm/385567299710?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=51dshkl2rvy&sssrc=2349624&ssuid=VNMmCGBET96&var=&widget_ver=artemis&media=COPY

ChrGri, 2023-04-25:
These DYLY-107 loadcells look similar to those used on the simucube pedals and the dimensions are reasonably small

https://a.aliexpress.com/_EQln9jz

jgbbob8, 2023-04-25:
I found a tiny one. Yo can check it here https://github.com/jgbbob8/DIY-Sim-Racing-Active-Pedal

kiss81, 2023-04-26:
> I found a tiny one. Yo can check it here https://github.com/jgbbob8/DIY-Sim-Racing-Active-Pedal

I like how small it is. You do need a relative short arm to get a good resolution. Your design seems to cope with that. The downside will be a lot of high force and I am not sure if you're motor will be strong enough to handle it. I do think a longer arm and ~100kg loadcell  will work better to be honest.

tjfenwick, 2023-04-26:
I tried the ads1115 and found the resolution to be not fine enough to smoothly work. Resolution with the 300kg load cell end up only being 0.1kg which lead to the pedal moving in steps rather than a smooth motion. The main problem is the ads1115 doesn't have a large enough gain value. I am going back to the ADS1256 and trying a linear voltage regulator to keep a consistent 5v input.

ChrGri, 2023-04-26:
@tjfenwick in your video, one can see some oscillations when the pedal is ac-/decelerated. Due to this acceleration additional force (F=m*a) will be measured by the loadcell. Do the oscillations you observed also occur, when you decouple the loadcell from the motor, thus the loadcell is not moving at all?

tjfenwick, 2023-04-26:
Yes, the fluctuations occur even when the load cell is completely still. I believe it has to do with the 5v supply from the Arduino flucuating. Just a 0.01mV fluctuation can cause the load cell to read a whole kilogram different. This problem doesn't occur when you have a low sample rate (less than 100 samples per second) since it can average a bunch of measurements. However when you get more and more above 100 sps the fluctuations get worse.

jgbbob8, 2023-04-26:
@tjfenwick  may be a 300kg load cell is to much, I use a 150Kg one and do not press more tahn 75kg.  I thnonk that If I make 75kg presure my range must be about 0 to 50% in a 300 KG cell the range will be 0 to 25% if I am not wrong

ChrGri, 2023-04-26:
@tjfenwick I would have thought, that due to the differential operation of the ADC, reference voltage fluctuations would not have an impact of the loadcell readings. But I'm gonna check the TI documents during the weekend.

tjfenwick, 2023-04-27:
That's what I thought but figured I'd give the linear voltage regulator a try. 

The voltage regulator seemed to help as it cut the average fluctuations were cut in half. However I can still see voltage changes on my voltmeter especially when wiggling wires in the breadboard. When wiggling the input voltage wires to the ADS1256 the fluctuations get much worse.

Perhaps using a solderable breadboard would further help with this issue.

jjzurro, 2023-04-28:
Hi, 
there are many load cell brake implementations that uses the INA122, it looks very fit for this task..

bbbkada, 2023-04-29:
I use 2 HX711 with Teensy Lc for my loadcells (brakepedal, handbrake), it's easy and doing the job.

ChrGri, 2023-04-30:
> I use 2 HX711 with Teensy Lc for my loadcells (brakepedal, handbrake), it's easy and doing the job.
> 
> https://www.sparkfun.com/products/13879

Thanks for the suggestions. I've used the HX711 successfully in passive loadcell pedals. However, the HX711 only supports up to 80SPS. Tried 100SPS with the ADS1256, which resulted in strong oscillations.

ChrGri, 2023-04-30:
@tjfenwick: I've quickly tried an exponential filter, which gives faster response compared to the MA filter. I've made a video of the force readings during some pedal movement. Blue line is the noisy sensor reading, yellow is the MA filter (length=40) and gray is the exp filter (alpha=0.15). The oscillations aren't noticeably worse than with the MA filter, but the lag is reduced significantly.
https://user-images.githubusercontent.com/21274895/235343621-2207ca0e-c85c-4bd4-90c4-0a3b1a32e098.mov

ChrGri, 2023-04-30:
@tjfenwick: Same test, but with a Kalman Filter implementation now. 

Blue: instantaneous force measurement
Gray: exp filter output
Yellow: Kalman observation

https://user-images.githubusercontent.com/21274895/235350153-70ea7c90-e059-40ad-acf6-2662aa5a47d3.mov

tjfenwick, 2023-04-30:
Awesome work! I'll have to try both of those filter methods and see how they work for me. 

In those video are you using a ADS1256 and what was the sample per second?

ChrGri, 2023-04-30:
> Awesome work! I'll have to try both of those filter methods and see how they work for me.
> 
> In those video are you using a ADS1256 and what was the sample per second?

Thank you. Both filters have been uploaded to the fork. ADS1256 was used with 500SPS. Let me know if you need assistance with the tuning. I've tried 1000SPS as well, but the cycle time on the Arduino Nano didn't improve over 500SPS, therefore didn't make sense.

SfcoAwol, 2023-05-10:
hey, I'm an idiot so feel free to ignore this if its not helpful and there might be reason that you haven't done this but when i experienced this problem using potentiometers i was getting spikes that would spike over half the sweep of my pedal and i could never create a large enough deadzone to actually manage it and i found that simply placing a capacitor from the ground to your analog in pin would make all the difference, now I'm sure doing this might introduce a little latency but i would guess that could be minimalized based on the size of cap that you choose

ChrGri, 2023-05-11:
Hi, my pedals are almost ready to go to the sim rig. Only waiting for the custom PCB to be delivered. 
The 300kg loadcell works perfect with a shortened cable. I don't notice any pedal oscillation or lag with the proposed Kalman filter. 

The closed-loop stepper has been tuned as shown below:
[image]

Pe0.03: 11 --> 9
Pe0.04: 600 --> 100
Pr0.08: 1600 -> 500

Subjectively speaking, the pedal feel is much smoother and the motor quiter with these changes.

laserbeak, 2023-05-18:
> Subjectively speaking, the pedal feel is much smoother and the motor quiter with these changes.
I've had a look at your fork. Great work. How's testing?

jgbbob8, 2023-05-18:
When will be able the wiring diagram? I do not want to burn anything jajajajaja

ChrGri, 2023-05-18:
@laserbeak @jgbbob8 I've designed and ordered custom PCBs for the DIY pedals. They should come early next week. It was 8€/5PCBs including shipping. If you want, I can release the Gerber files after testing?

[image]

laserbeak, 2023-05-18:
> @laserbeak @jgbbob8 Thank you. I've designed and ordered custom PCBs for the DIY pedals. They should come early next week. It was 8€/5PCBs including shipping. If you want, I can release the Gerber files after testing?
> 

That's very generous of you and yes, I'd be interested. I'm also wondering how your solution mounts to an 80/20 rig. My current fanatec pedals (Clubsport) are directly mounted to mine. I can see myself mounting them with corner brackets, but would still need a footplate to rest my heels.

speedcrafter442, 2023-05-29:
> @laserbeak @jgbbob8 Thank you. I've designed and ordered custom PCBs for the DIY pedals. They should come early next week. It was 8€/5PCBs including shipping. If you want, I can release the Gerber files after testing?
> 
> [image]

Is there somewere a schematic for this PCB?

ChrGri, 2023-05-30:
https://github.com/ChrGri/DIY-Sim-Racing-Active-Pedal/tree/main/Wiring