tjfenwick/DIY-Sim-Racing-Active-Pedal #2: Hardware design suggestion

ChrGri, 2023-04-23:
Thanks @tjfenwick for his awesome design. To support this project, let's collect and discuss some hardware realizations to potentially improve the original design.

Loadcell rod:
The original design is 3D printed and I'm not sure how it will last over time. 
Since the loadcell already has an M12 thread. Use a short M12 threaded rod on both sides of the load cell. At the end of these threaded rods, use M12 ball joints, e.g.
https://amzn.eu/d/eeJQwHZ
to replace the 3D print and ball bearings.

Body, guides, and lead screw:
The original design is 3d printed.
Of the shelf modules for CNC machines might be applicable here, see e.g.
https://a.aliexpress.com/_Ez9AyIL

Pedal:
Use 2020 extrusions, e.g. like on this pedal set
https://www.najam-simtec.de/i/k2x-pedals

BR
Christopher

ChrGri, 2023-04-24:
I've made a construction from mostly metal norm parts. Parts are already ordered. Going to test it soon.
[image]

jjzurro, 2023-04-26:
Hi guys, 
I really like your project. I don't know how far it can go (car profile simulation, abs simulation, etc), but I love it!!!
I'm a total noob, but maybe it will be a better result using a bldc motor with and esc controller. there is a lot of videos about the fast response, readings precision and torque.

I hope it can help you

kiss81, 2023-04-26:
Love the design! Looks pretty strong! Any chance you can put a list with the components / sizes?

tjfenwick, 2023-04-26:
Right now everything is constantly changing since this is still a very early prototype. Once I get a more finalized design I will put together a bill of materials with links on were to buy everything.

jgbbob8, 2023-04-28:
> Once the electronics and software are done, smaller components will be tested. [image]

Very nice, and well resolved the size of the motor under the pedal in the area of ​​the heels... great idea

jjzurro, 2023-05-02:
Hi everyone!
I love this project and I want to help and build it!!!

-Did you though about including an eeprom for the project for custom profiles? (or maybe a stm32, that is faster  than arduino , also have HID library and have more of everything), I think it's the final goal of having those kind of pedals, to be able to load a specific profile for a specific car....

-I found this, it could be a good cheap option:

[https://www.aliexpress.com/item/32955790850.html?spm=a2g0o.productlist.main.111.60dd59a7hQWP1u&algo_pvid=e5d30f50-8522-45b6-a288-45020817f6b7&algo_exp_id=e5d30f50-8522-45b6-a288-45020817f6b7-55&pdp_npi=3%40dis%21JPY%212704.0%212704.0%21%21%21%21%21%40214527fd16830000706746729d07b7%2112000023008795451%21sea%21JP%211775284287&curPageLogUid=z44XgnDlQQIp](url)

or this [https://www.aliexpress.com/item/1005004022886871.html?spm=a2g0o.detail.0.0.3c51b54cbxAABj&gps-id=pcDetailFavMayLike&scm=1007.14452.331081.0&scm_id=1007.14452.331081.0&scm-url=1007.14452.331081.0&pvid=f8129fcf-567c-476d-aa4f-3bb2ed79967b&_t=gps-id:pcDetailFavMayLike,scm-url:1007.14452.331081.0,pvid:f8129fcf-567c-476d-aa4f-3bb2ed79967b,tpp_buckets:668%232846%238110%231995&pdp_npi=3%40dis%21JPY%215672.0%213913.0%21%21%21%21%21%40210324e516830010253947058eb333%2112000027776359282%21rec%21JP%211775284287](url)

Also, it could be a very good idea to have  a discord channel to discuss all that stuff. Have a nice day!!

AyrtonRicardo, 2023-05-04:
> Once the electronics and software are done, smaller components will be tested. [image]

@ChrGri mind sharing the dimensions of the profiles? I do happen to have enough alu profile near me and I want to try out the design.

bbbkada, 2023-05-04:
They have a 3d .step file for download on their wesite, if someone wants inspiration. by the size, of the motor, I guess they have a 130w ?! My uploaded file online Autodesk view, will be up for 30 days if you want a quick peek.
https://autode.sk/3NANJBC

ChrGri, 2023-05-05:
@AyrtonRicardo Sure, I've used 2020 V-Slot profiles, which I had laying around as well. The CAD model has been uploaded to the fork:
https://github.com/ChrGri/DIY-Sim-Racing-Active-Pedal/tree/main/CAD/FromNormParts_V1

However, this is how the current design looks like:
[image]

These ready-to-go CNC modules are surprisingly cheap.

ChrGri, 2023-05-26:
So I've tested the initial version of my custom PCB design for the ESP32 S2 Mini and it's working beautiful. Here is a picture of it mounted to the pedal:

[image]

kiss81, 2023-05-26:
> So I've tested the initial version of my custom PCB design for the ESP32 S2 Mini and it's working beautiful. Here is a picture of it mounted to the pedal:
> 
> [image]
> 
> It's going to the rig next.

That's looking really nice! You already got it working as 'joystick' input device as well? A video would be really nice to see it working. Might copy the whole design and build one myself!

kiss81, 2023-05-26:
That;s serious progress @ChrGri ! Looking forward to the video!

ChrGri, 2023-05-26:
@kiss81 Thank you :) Here is a screenshot showing the joystick output from the pedal and a video showing it in action.

[image]

https://github.com/tjfenwick/DIY-Sim-Racing-Active-Pedal/assets/21274895/4f2327b7-4ec6-4d88-8ae2-2e65f62bca9d

Ibakha, 2023-05-26:
I don't know if the necessary torque was calculated. I was planning on looking at it this weekend. But if you can place the maximum course in the middle. Is this motor strong enough to hold in place and possibly fight back if the abs were implemented?

kiss81, 2023-05-27:
@ChrGri that looks so useable allready! Anything that needs more improvements? (except things like a pedal plate and configuration panel)
@Ibakha ABS feedback would be really nice. Not sure how that feels / what happens in a real race car if it kicks in?

ChrGri, 2023-05-27:
@kiss81 I drove iRacing yesterday for an hour. Pedal felt awesome, more alive than the passive pedal I have, even though no ABS has been implemented.
Currently the pedal seems to skip some steps and loses its neutral position. So after some driving the pedal tilts more and more towards the ground. I will try to find the reason for this next.

kiss81, 2023-05-27:
So cool to hear! Hopefully the motor is not too weak... Any idea how much pressure you need now to hit the brakes 100%?

Ibakha, 2023-05-27:
@ChrGri Do you have an updated version of your CAD model? I am modelling the pedal with [these](https://www.aliexpress.com/item/1005004876489894.html?spm=a2g0o.productlist.main.113.619a7c11pSXw0Y&algo_pvid=6bf062e2-b598-4df6-b6b2-9de94cd7629e&algo_exp_id=6bf062e2-b598-4df6-b6b2-9de94cd7629e-58&pdp_npi=3%40dis%21CAD%21214.92%21137.55%21%21%21%21%21%402102111816852074929757165d07ee%2112000030856413660%21sea%21CA%212601918725&curPageLogUid=zhw08wE1X1RU), which I think you are using.
I'll have someone looking at my math this week, but I'll try to upload an Excel to calculate the minimum torque needed depending on the geometry of the pedals. Maybe I'm worrying for nothing, but it annoys me.

@kiss81 I don't know if it's different in a race car, but I'm pretty sure the reaction of the pedals is similar to your day to day car. Where the brake pedals is pushing back as the ABS releases the brakes.

kiss81, 2023-05-27:
@ChrGri @Ibakha I just had the opportunity today to drive the simucube active pedals on brake and throttle. The ABS indeed felt really like a road car. You could feel the ABS ON/OFF to keep the tires rotating while the ABS was active. The Throttle felt kind of nice too, but hard to explain what they did actually. Need more time with that.

ChrGri, 2023-05-28:
@Ibakha You can find the CAD model of my previous version here https://github.com/ChrGri/DIY-Sim-Racing-Active-Pedal/tree/main/CAD/FromNormParts_V1. But I don't have an updated CAD model with the CNC module. I've ordered this (SFU1605, 100mm travel), but it should be identical to the ones from other sellers:

https://www.ebay.de/itm/393149454745?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=rgcaefgvrgm&sssrc=2349624&ssuid=6twGLuBEQmS&var=662086958084&widget_ver=artemis&media=COPY

Regarding the force, it's always good to have some kind of analytical model to verify the selected parts. Keep in mind, that you do not really need to apply a lot of torque to block the spindle from turning and therefore stop the pedal movement.