tjfenwick/DIY-Sim-Racing-Active-Pedal #4: some ideas: cheaper motor + electronics

kiss81, 2023-04-26:
I see you have plans to go for a teensie 4.0 for better performance. I do think you want a teensie 4.1 as it can act as USB peripheral (useful for make it an actual controller).

But the whole project is kind of expensive for a DIY if you want to make a three pedal setup for example. Some ideas:

pick a esp32 board as they are cheap and have lot's of performance for < 10$

tmc2209 stepper driver < 10$
https://www.amazon.com/BIGTREETECH-DIRECT-TMC2209-Stepsticks-Motherboard/dp/B07ZQ3C1XW

a nema23 motor
https://www.amazon.de/-/en/23HS9440-26-Stepper-Bipolar-D-shaft-1-8%C2%B0CNC/dp/B084RPGHXH
For the motor I think you can give some advice how many RPM / force is needed. Do you have any idea?

software to use it as 'controller':
https://github.com/lemmingDev/ESP32-BLE-Gamepad
If this doesn't work or gives too much latency I might add a USB joystick controller

I have plans to start experimenting as well, but will start with the motor / electronics / software first.
Would love to hear your thoughts.

GManzato, 2023-04-27:
An open loop stepper motor would make things more complex, you would have to min/max at each start and ensure to have powerful enough to not skip step. I think it's not worth.

tjfenwick, 2023-04-27:
As for the micro controller I decided against the teensy 4.0/4.1 since the stepper motor library needed for high speed movement and acceleration doesn't exist at the moment and I am not skilled enough to write one.

I have switched to an Arduino Pro Micro since it can communicate via USB like a controller. I also decreased the steps per revolution on the motor to 500 to decrease the steps per second required to drive the motor.

ChrGri, 2023-04-27:
@tjfenwick could you post a wiring diagram or photo of your setup for reference?

tjfenwick, 2023-04-27:
Yeah, I'll work on getting one together to post along with updated code.

kiss81, 2023-04-27:
> The other issue with a regular stepper motor is the lack of higher speed torque. Most stepper motors seem to max out around 1600 rmp and lose torque drastically after 1000 rpm. The integrated servo motor I used will spin up to 4000 rpm and hold a decent amount of torque.
> 
> You need a high speed motor 2000+rpm other wise your pedal will move to slowly.

I have no experience with integrated servo motors but I do share your concerns on the torque. I don't think we need that much RPM's as the spindles have 2mm pitch. So for like 100mm movement you do need 100rpm to move full in/out in half a second...

ChrGri, 2023-04-28:
@kiss81 I've had a 3D printer with a NEMA17 and TR8 spindle on the z-axis. As far as I remember, the z-axis top speed was approx. 20mm/s before stalling. This was with a TR8 spindle and only a couple of 100 grams to move. 

For the application of the active pedals, I roughly guess, that 100mm/100ms are required = 1000mm/s. Where the 100mm comes from the expected travel range of the spindle and the 100ms is the time to completely push the pedal. 

From my past experience I am certain, that open loop steppers wouldn't be sufficient. But when it comes to decrease the pricing of this project even further, I would be happy to be proven wrong.

tjfenwick, 2023-04-28:
When calculating how fast your motor needs to be you need to think about how fast you can fully compress your current gas, brake or clutch pedal. I estimated it to be around 0.1-0.2 seconds. To achieve this assuming your max travel distance was 50mm and your ball screw pitch was 5mm/rev you need your motor spinning at 4000rpm. Also remember your motor doesn't start at 4000rpm it will have to accelerate to that speed.

t=(Travel/Pitch)/(RPM/60)

B3N1321, 2024-04-08:
can this design be compatible with this Nema 23 motor?

Just found this amazing item on AliExpress. Check it out! HUF12,788.29  50％ Off | Nema23 Closed Loop Integrated Motor 1.2Nm 2Nm 2.5Nm 3Nm Diameter 8mm Hybrid Integrated Servo Motor with Drive For CNC
https://a.aliexpress.com/_EQKCyEF