tjfenwick/DIY-Sim-Racing-Active-Pedal #9: Discord - tracking todos

Ibakha, 2023-05-28:
Is opening a discord server or starting the project to track the todos & what's in progress?

ChrGri, 2023-05-30:
I like Discord, but Discord servers are not for free, are they?

Ibakha, 2023-05-30:
They are, you can unlock some features by paying, but they are not necessary.

Ibakha, 2023-05-30:
Here the invitation for the discord, I'll add moderators and transfer the ownership has people join. I used a template, so I'm open to suggestions to update the server format.

https://discord.gg/qpVkXbxRgA

Ibakha, 2023-05-30:
@tjfenwick, is it possible for you to create webhooks for the project? I'd like to add notifications to the discord for pull requests & releases

jgbbob8, 2023-05-30:
> 

Sorry, Discord INvitation not valid...

Ibakha, 2023-05-31:
@jgbbob8 here's the invitation. I forgot to remove the expiration, sorry
https://discord.gg/kzYXykdJRT