tjfenwick/DIY-Sim-Racing-Active-Pedal #12: Wiring Questions and Problem programming Pro Micro

jman-31, 2023-08-28:
Hey there, 

I am working on a design similar to yours and would love to know the wiring you used. Would it be possible to get a photo of your wiring or have you had time to make a diagram? Also, I am pretty good at designing circuits and PCB's so I would be glad to throw my hat in the ring to help with this project. I have attached a photo of my pedal so far. It's going to be belt driven as opposed to direct drive just to make it a smaller package. Still waiting on the belt, gears and a couple of bearings to give it a test, but you are way ahead of me on programming part so I didn't figure to reinvent the wheel.

Also, I tried to upload your code to a Pro Micro and I got the error:
#error "Oops! Your board architecture is not supported yet'" (from the ADS1256 library)
Have you tried this with a Pro Micro? I looked through the ADS1256 library and I didn't see support for the ATmega32u4 processor. I could very well be doing something wrong. Thanks for any advice.

My email is [jer_ulm@yahoo.com](mailto:jer_ulm@yahoo.com) if you want to contact me.

[image]

ChrGri, 2023-08-29:
Great design :)
The discord server, mentioned in another issue, has a lot of active users and is a good place for discussion.

jman-31, 2023-08-29:
Thanks! I appreciate the info. I visited the discord and it looks like I've got a lot of catching up to do!!